UWCubeSat/DubSat1 #3: Make #define-based mechanism for specifying which I2C module is being used on MSP43x

jeffchrisope, 2017-04-29:
_From @jeffchrisope on April 10, 2017 7:33_

_Copied from original issue: UWCubeSat/I2C#6_

jeffchrisope, 2017-04-29:
Done.

jeffchrisope, 2017-04-29:
Oops, closed the wrong one.  This is still very much alive, but "nice to have" at this point, so low priority.

jeffchrisope, 2017-05-09:
New and maybe unfortunate requirement:  we MIGHT have to support multiple I2C buses going to the same 430, particularly in the ADCS subsystem, which has lots of "submodules" that will need to talk and one I2C for all of it is probably not the right call.

jeffchrisope, 2017-05-13:
See #39 for other possible twists on this - particularly the need for MULTIPLE peripherals ... ugh.

HuskySatLab, 2017-10-26:
Multiple buses now supported (as well as multiple devices.)

closed 2017-10-26 by HuskySatLab