UWCubeSat/DubSat1 #65: Known new CAN messages needed

jeffchrisope, 2017-08-20:
Please use this issue/item as a temporary storage place for the CAN messages we have discussed recently, that need to be included in the official DBC file once that is picked up by the appropriate person.

Definitely needed:
- "Wake up" message:  sent upon a subsystem starting up, with as much info from registers as possible to indicate why they were power cycled
- "Request to power cycle" - since our power domains won't be fine-grained enough to allow an MSP to cycle individual (or all of its) sensors, we are going with the model that they can either power cycle just their chip (typically done by writing to the watchdog register without the magic password value, hence forcing an MSP 'reboot') or a true power cycle, by sending out this message.  The power system would listen for it, and power cycle the requesting board's power entirely.

djdupre, 2017-09-14:
I'v proposed some GPS messages in this commit: 02ffb1dfffb547e120a29692e1cb78381083c7be.

I'll see if these the Rain Lab guys agree.

djdupre, 2017-09-15:
I want to broadcast six floats for standard deviation of position and velocity. Should they each have their own message, or should they be grouped to fit in just three messages?

jeffchrisope, 2017-09-15:
Pack ‘em in as tightly as you can, the more the merrier.