UWCubeSat/DubSat1 #71: CORE:  Timer library v0.9

HuskySatLab, 2017-10-28:
This initial version will have core timer functionality, allowing for easy, low-overhead timer tracking.

MrWilson1, 2017-10-28:
Task 1:  Design and Implement simple timers:
                a) Compile the existing Core code onto an MSP5994 board and become familiar BSP.h
                b) Using TimerA (values from 1uS to approx 65ms).......
                     1) Create a setTimerToZero_us_resolution() function that returns the value of TimerA
                            - Application defines variables, say CAN_timeout and I2C_timeout
                            - "clear" these variables as follows - in the appropriate places in the app code
                                     CAN_timeout = setTimerToZero_us_resolution():  //turn ON LED 0
                                     I2C_timeout = setTimerToZero_us_resolution():  // turn ON LED 1
                     2) Then use chkTimer_us_resolution(int timer_value) 
                             - Application would use in appropriate places in App code
                                      if(chkTimer_us_resolution(CAN_timeout) > 50ms //make this a define
                                      {
                                            do something here - maybe record an error
                                            For now, turn OFF the LED 0 //put in place holder for now - wait for d) below to get this working
                                      }
                            
                                      if(chkTimer_us_resolution(I2C_timeout) > 2ms //make this a define
                                      {
                                            do something here
                                            For now, turn OFF the LED 1 //put in place holder for now - wait for d) below to get this working
                                      }
                b) Using variable "seconds_timer" (values from 65ms to approx 72 minutes)......
                     1) Use a setTimerToZero_sec_resolution() function that returns the value of seconds_timer
                            - Application defines variable, say Sync_timeout
                            - "clear" as follows - in appropriate place in the app code
                                     Sync_timeout = setTimerToZero_sec_resolution():  //turn ON LED 3
                     2) Then use chkTimer_sec_resolution(int timer_value) 
                             - Application would use in appropriate place in App code
                                      if(chkTimer_sec_resolution(Sync_timeout) > 800ms //make this a define
                                      {
                                            do something here - maybe record an error
                                            For now, turn OFF the LED 2  //put in place holder for now - wait for d) below to get this working
                                      }

                  d) create test functions that use LED 0, 1 and 2 in bsp.h - make sure we do not check
                        these into main

MrWilson1, 2017-10-31:
Timer Type 2:  Design and Implement:

1. An Event Timer (with a 10ms Interrupt Rate, for now) Requires
    - An Interrupt Service Routine
       - Code to reload the event timer overflow register (each time)
       - 2 lines of code for each timer
    - Define 10 general timers, which will count down to zero (general_timer1, general_timer2, etc)
    - A set_timer(timer name, timer value) function that
       - Relates timer name to one of the general timers
          - Intent is to hide general timers from app developer 
          - So that the ISR does not need to be updated for each new App timer
    - A chk_timer_expired(timer name) function
       - that returns True if timer equals zero
       - Needs to remove timer name from list when done (free up general timer)