UWCubeSat/DubSat1 #90: ADCS:  Get autocode from SL running on a 432 and perform memory/perf tests

jeffchrisope, 2017-11-01:
This is the first real attempt to get library blocks from Simulink running on the MSP432 chip.

HuskySatLab, 2017-11-01:
Updating MSP432 toolset standard version in repo.

HuskySatLab, 2017-11-01:
[image]
Memory usage (see project src/devmods/adcsfswtest_cdr_msp432), deployed onto actual MSP432 Launchpad.  rt_OneStep() call rates in the 35-45Hz region, which is a good starting point - though note that no actual sensor data is being fed into the system, so things will likely slow when there are more "things to do."

jeffchrisope, 2017-11-03:
Actually ... new process:  tasks aren't closed until the sprint meeting, so they can be confirmed done.  The task cards can (and should) still be moved to the completed column once done, however.

closed 2017-11-11 by HuskySatLab