UWCubeSat/DubSat1 #184: Distribution board wrecks CAN bus

garrettdreyfus, 2018-05-25:
Busmaster showed just bit errors being sent. We narrowed it down to the distribution board by turning off every power domain and it still sent errors.  The SPI line on the bus was acting very strangely

[image]
[image]
[image]
[image]
[image]

emorye, 2018-05-25:
How is the spi line acting weird?

garrettdreyfus, 2018-05-25:
Those photos of the oscilloscope are of the SPI line.

emorye, 2018-05-25:
Also is it reproducible? What are the steps to reproduce? @Paul-Sturmer @garrettdreyfus

garrettdreyfus, 2018-05-25:
The COM2 board was attached also not sure if it caused the FlatSat to be put in a bad state.

emorye, 2018-05-25:
Looks like the spi line is acting normally, characteristic of 3 tx buffers full I'll have to check why it thinks all the buffers are full.

garrettdreyfus, 2018-05-25:
That is good to know

garrettdreyfus, 2018-05-25:
Emory told me that "It looks clear that we MUST check what canSendPacket() returns
because I can 99% guarantee you it is returning nonzero (for Send failed)
(for boards in general)"  So maybe that should be the criteria for the distribution board to reset itself? @nkwacker  @Paul-Sturmer @jeffchrisope

garrettdreyfus, 2018-05-25:
Paul says this is due to deployment switch.

closed 2018-05-25 by garrettdreyfus

emorye, 2018-05-25:
What's the deployment switch?

Paul-Sturmer, 2018-05-25:
it's the switch the when depressed unpowers the entire satellite. this is
how the satellite turns on when it leaves the dispenser