UWCubeSat/DubSat1 #185: MET lengths are reversed in dist rc

nkwacker, 2018-05-26:
[image]
Note how overflow is 4 bytes where it should be 1 and met is 1 byte where it should be 4.

Also on a slightly unrelated note, in the COSMOS packet `RC_EPS_DIST_2` only has `RC_EPS_DIST_2_MET` but not the overflow field

garrettdreyfus, 2018-05-26:
> Assigned #185 <https://github.com/UWCubeSat/DubSat1/issues/185> to
> @garrettdreyfus <https://github.com/garrettdreyfus>.

:P. Just switched the two.

nkwacker, 2018-05-26:
What about the COSMOS issue?

garrettdreyfus, 2018-05-26:
canwrap issue resolved

emorye, 2018-05-26:
(Reopened b/c Cosmos issue not resolved)

nkwacker, 2018-06-14:
Remaining problem appears in issue #201

closed 2018-06-14 by nkwacker